LiteOS/LiteOS #25: Some views about the using of `__cpluscplus`

cifer-lee, 2017-01-17:
I found that most of the kernel source files using following grammar:

```
#ifdef __cplusplus
#if __cplusplus
extern "C" {
....
}
#endif
#endif
```

like here: [https://github.com/LITEOS/LiteOS_Kernel/blob/master/kernel/base/mem/los_memory.c#L43](https://github.com/LITEOS/LiteOS_Kernel/blob/master/kernel/base/mem/los_memory.c#L43)

I think the inner `#if __cplusplus` is not needed here, the outer `#ifdef __cplusplus` can handle things properly. Actually, there exists quite a few old compilers will define `__cplusplus` to 0 or just empty, which would cause the wrapped source code not be compiled at all.

Do we have other considerations of using two overlapped `#ifdef __cplusplus  #if __cplusplus` here?

dndxhej, 2017-04-10:
#ifdef __cplusplus is enough, and #if __cplusplus seems useless.
And in c++03, __cplusplus is 199711L, and in C++11, __cplusplus is 201103L. If the compiler version is newer, #if __cplusplus will not cause any issue.

BaikalHu, 2017-04-19:
And in c++03, __cplusplus is 199711L, and in C++11, __cplusplus is 201103L.
In ubuntu 16.04,  we could find in /usr/include/x86_64-linux-gnu/c++/5/bits/c++config.h:# if __cplusplus >= 201103L, so #if __cplusplus won't cause any issue.

closed 2017-04-19 by BaikalHu