LiteOS/LiteOS #39: 动态内存释放失败问题(参考提供的例子)

simplezhao, 2017-03-27:
### 动态内存释放失败
参看动态内存释放的例子，编写如下测试程序，内存初始化、内存申请都成功，但是内存释放却失败了。没看出哪里有问题，代码如下:
(usart2_printf是一个串口打印字符串的函数，硬件基于STM32f103RBT6平台，下面这个函数是放在一个任务中执行的)

```C
VOID los_memory_test() {
    char *p_num = NULL;
    UINT32 uwRet;
    uwRet = LOS_MemInit(m_aucSysMem0, 1024);
    if (LOS_OK  == uwRet) {
        usart2_printf("mempool init ok!\r\n");
    }
    else {
        usart2_printf("mempool init failed!\r\n");
        return;
    }
    /*申请内存*/
    p_num = (char*)LOS_MemAlloc(m_aucSysMem0, 8);
    if (NULL == p_num) {
        usart2_printf("mem alloc failed!\r\n");
        return;
    }
    usart2_printf("mem alloc ok\\r\n");
    /*内存赋值*/
    p_num = "hello";
    usart2_printf("*p_num = %s\r\n", p_num);
    /*释放内存*/
    uwRet = LOS_MemFree(m_aucSysMem0, (char*)p_num);
    if (LOS_OK == uwRet) {
        usart2_printf("mem free ok!\r\n");
    }
    else {
        usart2_printf("mem free failed:%d\r\n",uwRet);
    }
    return;
}
```
**执行到室内内存时，uwRet = 1**

liuxuming, 2017-03-28:
/*内存赋值*/
    p_num = "hello";
    usart2_printf("*p_num = %s\r\n", p_num);
    /*释放内存*/
    uwRet = LOS_MemFree(m_aucSysMem0, (char*)p_num);

你释放的是栈中的p_num.

simplezhao, 2017-03-28:
我有新的问题，使用动态内存时，申请的内存(8*char)小于赋值的的内存(“hello world”)，正常输出，但是内存释放却失败了；实际使用内存小于申请的内存时，可以成功释放。
```C
VOID los_memory_test() {
    char *p_num = NULL;
    UINT32 uwRet;
    char* str = "hello world";
    int i;
    uwRet = LOS_MemInit(m_aucSysMem0, 1024);
    if (LOS_OK  == uwRet) {
        usart2_printf("mempool init ok!\r\n");
    }
    else {
        usart2_printf("mempool init failed!\r\n");
        return;
    }
    /*申请内存*/
    p_num = (char*)LOS_MemAlloc(m_aucSysMem0, 8);
    if (NULL == p_num) {
        usart2_printf("mem alloc failed!\r\n");
        return;
    }
    usart2_printf("mem alloc ok\\r\n");
    /*内存赋值*/
    for(i = 0; i < strlen(str); i++){
        *(p_num + i) = str[i];
    }
    usart2_printf("*p_num = %s\r\n", p_num);
    /*释放内存*/
    uwRet = LOS_MemFree(m_aucSysMem0, (char*)p_num);
    if (LOS_OK == uwRet) {
        usart2_printf("mem free ok!\r\n");
        p_num = NULL;
    }
    else {
        usart2_printf("mem free failed:%d\r\n",uwRet);
    }
    return;
}
```

zhoubo85, 2017-03-31:
for(i = 0; i < strlen(str); i++){
        *(p_num + i) = str[i];
    } 这段代码，使用的内存大于8，这个将导致数据破坏内核中的内存管理链表(所以会释放失败)。所以请使用者自行保证内存使用不能越界。

closed 2017-04-01 by liuxuming