LiteOS/LiteOS #46: Code Style is not so good.

larzhang2, 2017-04-01:
I have just reviewed usr/main.c and found that there were a few meaningless blank space in the end of line and tab space in the beginning of line. I suggest you guys could use vim plug-in unit for detecting this code style issue. 

As Huawei is an influent IT company, I think it's better to make the source codes to look beautiful.

[image]

liuxuming, 2017-04-01:
thanks, we'll improve immediately.

larzhang2, 2017-04-01:
https://github.com/google/styleguide/tree/gh-pages/cpplint
Maybe this tool is helpful for you guys.

closed 2017-04-06 by liuxuming