LiteOS/LiteOS #49: 关于liteos使用中断的疑问，感觉自己在用‘假系统‘

simplezhao, 2017-04-08:
按照参考文档编写下面的代码：
* 1、中断处理函数
```C
void EXTI15_10_IRQHandler(void){
        usart2_printf(" get in key int0\r\n");
        if(EXTI_GetITStatus(EXTI_Line13) != RESET){
	        GPIO_SetBits(GPIOA, GPIO_Pin_5);
        }
        EXTI_ClearITPendingBit(EXTI_Line13);
        return;
}
```
* 2、在liteos创建中断服务
```C
void hwi_test()
{
        UINTPTR uvIntSave;
        UINT32 uwRet = 0;
         
    	key_init(); /*用户外部中断初始化*/
        uvIntSave = LOS_IntLock();
        uwRet = LOS_HwiCreate(40, 0,0,EXTI15_10_IRQHandler,0);
        if(uwRet != LOS_OK){
		      usart2_printf("create failed\r\n");
    	}
    	else{
		      usart2_printf("create succesful\r\n");
    	}
        LOS_IntRestore(uvIntSave);
        }
```
* 3、自己的中断初始化（功能正常，略）
* 4、第二步创建的hwi_test()，在osAppInit()使用
我的问题是：
    不进行上面第二步在liteos创建中断服务hwi_test()操作，中断（按键）可以用，进行第二步hwi_test()操作也正常用，那这什么区别吗？我对系统编程不太了解，感觉不出有什么优势，或者是自己的错误用法。即时我不小心把 LOS_HwiCreate中HwiNum填错了，它也照常运行（HwiNum正确的情况下，建立一个任务，调用函数LOS_HwiDelete(中断号)倒是真的能把中断个禁止了），还请指导。

oahjhao, 2017-04-09:
首先key_init是不是已经把中断回调和中断链接了？外部事件-中断号-回调，看看是不是已经有通路了？其他使用是正确的；
两者区别相当于买黄牛票和买正经票（前提都是真票），用户本身没感觉出票的差别是正常的，座位号在合理范围内变化也没关系（中断号就是的座位，不过有的你不能坐），对于售票员就不一样了，售票员需要把座位的目的地做个管理，控制走和停（rtos高中断负载下最长不能响应中断的时间越短，实时性越好），这样旅途才一切顺利。
不知道这么解释合适不：）

发自网易邮箱大师
在2017年04月09日 01:33，simple 写道： 按照参考文档编写下面的代码：

1、中断处理函数

void EXTI15_10_IRQHandler(void){
        usart2_printf(" get in key int0\r\n");
        if(EXTI_GetITStatus(EXTI_Line13) != RESET){
	        GPIO_SetBits(GPIOA, GPIO_Pin_5);
        }
        EXTI_ClearITPendingBit(EXTI_Line13);
        return;
}

2、在liteos创建中断服务

void hwi_test()
{
        UINTPTR uvIntSave;
        UINT32 uwRet = 0;
          
    	key_init(); /*用户外部中断初始化*/
        uvIntSave = LOS_IntLock();
        uwRet = LOS_HwiCreate(40, 0,0,EXTI15_10_IRQHandler,0);
        if(uwRet != LOS_OK){
		      usart2_printf("create failed\r\n");
    	}
    	else{
		      usart2_printf("create succesful\r\n");
    	}
        LOS_IntRestore(uvIntSave);
        }

3、自己的中断初始化（功能正常，略）
4、第二步创建的hwi_test()，在osAppInit()使用
我的问题是：
不进行上面第二步在liteos创建中断服务hwi_test()操作，中断（按键）可以用，进行第二步hwi_test()操作也正常用，那这什么区别吗？我对系统编程不太了解，感觉不出有什么优势，或者是自己的错误用法。即时我不小心把 LOS_HwiCreate中HwiNum填错了，它也照常运行（HwiNum正确的情况下，建立一个任务，调用函数LOS_HwiDelete(中断号)倒是真的能把中断个禁止了），还请指导。

simplezhao, 2017-04-09:
@oahjhao 非常感谢您的回到。
我是在初始化的时候对中断控制器做了操作，感觉像您说已经被建立了通路，我也试了一下，注释掉对中断控制器的操作，LOS_HwiCreate()是有效的，这应该是已经交给系统来控制了，对于您说的区别，例子很形象，我不是很理解，我还得好好体会一下 :smile:。

```C
void key_configuration(void)
{
	
         ...
         ...
	/* Configure EXTI0 line */
	EXTI_ClearITPendingBit(EXTI_Line13); 
	EXTI_InitStructure.EXTI_Line = EXTI_Line13;
	EXTI_InitStructure.EXTI_Mode = EXTI_Mode_Interrupt;
	EXTI_InitStructure.EXTI_Trigger = EXTI_Trigger_Falling;  
	EXTI_InitStructure.EXTI_LineCmd = ENABLE;
	EXTI_Init(&EXTI_InitStructure);
	/*下面的操作应该交给系统来处理*/
	/* Enable and set EXTI15_10 Interrupt to the lower priority */
	//NVIC_InitStructure.NVIC_IRQChannel = EXTI15_10_IRQn;
	//NVIC_InitStructure.NVIC_IRQChannelPreemptionPriority = 2;
	//NVIC_InitStructure.NVIC_IRQChannelSubPriority = 2;
	//NVIC_InitStructure.NVIC_IRQChannelCmd = ENABLE;
	//NVIC_Init(&NVIC_InitStructure);
}
```

lichao93, 2017-04-12:
不进行上面第二步在liteos创建中断服务hwi_test()操作，中断（按键）可以用，进行第二步hwi_test()操作也正常用，那这什么区别吗？
———之所以会出现这样的情况，是因为您工程中使用的启动汇编文件里面已经注册了中断向量表__Vectors，建立好了中断与中断处理函数之间的联系，所以不用LOS_HwiCreate进行注册也可以正常使用。
两者的区别在于：
（1）使用LOS_HwiCreate可以在中断使用前才进行注册，使用完后可使用LOS_HwiDelete删除，删除后还可以再次注册（如有需要还可以修改与中断号对应的中断服务函数），从而实现了系统中断的动态管理，在系统中断资源不够用的时候这种方式显得很有效。
（2）采用启动汇编文件进行中断注册的方式，一开始就注册了芯片支持的所有的中断（即使是系统中不使用的中断），之后无法编码进行修改。

closed 2017-05-02 by liuxuming