LiteOS/LiteOS #58: Hwi宏定义在readme.md中与代码不一致

larzhang2, 2017-04-19:
`OS_INCLUDE_HWI` and `OS_INCLUDE_HWI` mentioned in Readme.md both have changed, which are defined accordingly as `LOSCFG_PLATFORM_HWI` and `LOSCFG_PLATFORM_HWI_LIMIT` now. Please update um in line with source code.

lichao93, 2017-04-24:
OS_INCLUDE_HWI and OS_INCLUDE_HWI mentioned in Readme.md？I can't find it.

larzhang2, 2017-06-01:
sorry for mistake, they exist in HuaweiLiteOSKernelDevGuide.md
>编程示例
前提条件：
在los_config.h中，将OS_INCLUDE_HWI定义为YES。
在los_config.h中，设置最大硬中断个数OS_HWI_MAX_USED_NUM。
说明 目前的中断测试代码提供了基本框架，中断硬件初始化代码请用户根据开发板硬件情况在Example_Exti0_Init()函数中自行实现。 

But in los_config.h, I just could find#if defined (RAM_SIZE_LEVEL_0)
```
#define LOSCFG_PLATFORM_HWI                             NO
#elif defined(RAM_SIZE_LEVEL_1)
#define LOSCFG_PLATFORM_HWI                             YES
#elif defined(RAM_SIZE_LEVEL_2)
#define LOSCFG_PLATFORM_HWI                             YES
#else
#define LOSCFG_PLATFORM_HWI                             YES
#endif

/**
 * @ingroup los_config
 * Maximum number of used hardware interrupts, including Tick timer interrupts.
 */
#define LOSCFG_PLATFORM_HWI_LIMIT                       96
```

zhoubo85, 2017-06-02:
I'm sorry for that,we did some code optimization，so in source code we deleted the macro OS_HWI_MAX_USED_NUM and OS_INCLUDE_HWI, but forgot update HuaweiLiteOSKernelDevGuide.md.
we will fix this later

gustave-wang, 2018-06-01:
already fix. Thanks for you report.

closed 2018-06-01 by gustave-wang