LiteOS/LiteOS #65: Failed to build project STM32F429I_DISCO_GCC

tidyjiang8, 2017-04-28:
Hi, 

I'm so excited to see the gcc support in liteos, but when I tried to build STM32F429I_DISCO_GCC, I failed. 

I'm using the cross compiler toolchain: `arm-none-eabi-gcc(version 4.9.3)`

The log is attached as follow:
```
work@ubuntu:~/me/liteos/projects/STM32F429I_DISCO_GCC$ make
arm-none-eabi-gcc -g -O0 -Wall -mlittle-endian  -mcpu=cortex-m4 -march=armv7e-m -mthumb -mthumb-interwork  -mfloat-abi=hard -mfpu=fpv4-sp-d16 -D LOS_GCC_COMPILE -D _GCC_WRAP_STDINT_H  -ffreestanding   -D ARM_MATH_CM4 -std=gnu99  --specs=rdimon.specs  -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/base/include -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/include -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/link/gcc -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/config -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cpu/arm/cortex-m4 -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../platform/STM32F429I_DISCO -I /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../example/include -c  -o /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.o /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.c
In file included from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/link/gcc/cmsis_gcc.h:39:0,
                 from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.c:37:
/usr/include/newlib/sys/_stdint.h:31:19: error: conflicting types for 'int32_t'
 typedef __int32_t int32_t ;
                   ^
In file included from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.c:35:0:
/home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_os.h:161:13: note: previous declaration of 'int32_t' was here
 typedef int int32_t;
             ^
In file included from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/link/gcc/cmsis_gcc.h:39:0,
                 from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.c:37:
/usr/include/newlib/sys/_stdint.h:32:20: error: conflicting types for 'uint32_t'
 typedef __uint32_t uint32_t ;
                    ^
In file included from /home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.c:35:0:
/home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_os.h:160:22: note: previous declaration of 'uint32_t' was here
 typedef unsigned int uint32_t;
                      ^
Makefile:120: recipe for target '/home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.o' failed
make: *** [/home/work/me/liteos/projects/STM32F429I_DISCO_GCC/../../kernel/cmsis/cmsis_liteos.o] Error 1
work@ubuntu:~/me/liteos/projects/STM32F429I_DISCO_GCC$ STM32F429I_DISCO_GCC

```

Thanks & BR

zhoubo85, 2017-04-29:
First , thanks for your reply.when use different version of arm-none-eabi-gcc it doesn't work. some may can't find the header file. for example "can't find sys/_stdint.h", some may give info as "conflicting types for int32_t ......". we will improve to solve this promblem as soon as possible.

This time i give one way to sovle this problem  
1: modify /kernel/link/gcc/cmsis_gcc.h  #include <sys/_stdint.h>  to #include <stdint.h> 
2: modify env.mk
AR := arm-none-eabi-ar
CC := arm-none-eabi-gcc
NM := arm-none-eabi-nm
CPP := arm-none-eabi-cpp
OBJCOPY := arm-none-eabi-objcopy
OBJDUMP := arm-none-eabi-objdump

to: 
AR := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-ar
CC := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-gcc
NM := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-nm
CPP := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-cpp
OBJCOPY := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-objcopy
OBJDUMP := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-objdump
3: download arm gcc 4.8.3 and copy it to /opt/ and extract it . then do make , it will work.
 download link as bellow:
https://nj02all01.baidupcs.com/file/4869e6a6e1dc11ea0835e8b8213bb194?bkt=p3-14004869e6a6e1dc11ea0835e8b8213bb194cab557d0000003983fda&fid=100802510-250528-3669876813&time=1493438494&sign=FDTAXGERLBHS-DCb740ccc5511e5e8fedcff06b081203-mLJDYj23ezcTkBEco46Tl13KocU%3D&to=69&size=60309466&sta_dx=60309466&sta_cs=3&sta_ft=bz2&sta_ct=7&sta_mt=7&fm2=MH,Nanjing02,Netizen-anywhere,,guangdong,ct&newver=1&newfm=1&secfm=1&flow_ver=3&pkey=14004869e6a6e1dc11ea0835e8b8213bb194cab557d0000003983fda&sl=83034191&expires=8h&rt=sh&r=386249572&mlogid=2748374883349702626&vuk=282335&vbdid=105549401&fin=gcc-arm-none-eabi-4_8-2013q4-20131204-linux.tar.bz2&fn=gcc-arm-none-eabi-4_8-2013q4-20131204-linux.tar.bz2&rtype=1&iv=0&dp-logid=2748374883349702626&dp-callid=0.1.1&hps=1&csl=300&csign=jxOBRm%2B9AOkpJPmKw2etyaLT0P0%3D&by=themis

zhoubo85, 2017-04-29:
First , thanks for your reply. I found that when use different arm-none-eabi-gcc version, then make will meet some errors.some may give info as "can't find <sys/_stdint.h>", some like "conflicting types ....". we will solve this problem as soon as possible.

This time i will give a way to fixed this problem.
1: modify LiteOS_kernel/kernel/link/gcc/cmsis_gcc.h header include "#include <sys/_stdint.h>" to "#include <stdint.h>"
2: modify projects/STM32F429I_DISCO_GCC/env.mk

AR := arm-none-eabi-ar
CC := arm-none-eabi-gcc
NM := arm-none-eabi-nm
CPP := arm-none-eabi-cpp
OBJCOPY := arm-none-eabi-objcopy
OBJDUMP := arm-none-eabi-objdump

to 

AR := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-ar
CC := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-gcc
NM := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-nm
CPP := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-cpp
OBJCOPY := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-objcopy
OBJDUMP := /opt/gcc-arm-none-eabi-4_8-2013q4/bin/arm-none-eabi-objdump

3: download arm gcc 4.8.3  and extract it to /opt/ and then do make, it will work.
the download address is: 
https://nj02all01.baidupcs.com/file/4869e6a6e1dc11ea0835e8b8213bb194?bkt=p3-14004869e6a6e1dc11ea0835e8b8213bb194cab557d0000003983fda&fid=100802510-250528-3669876813&time=1493438494&sign=FDTAXGERLBHS-DCb740ccc5511e5e8fedcff06b081203-mLJDYj23ezcTkBEco46Tl13KocU%3D&to=69&size=60309466&sta_dx=60309466&sta_cs=3&sta_ft=bz2&sta_ct=7&sta_mt=7&fm2=MH,Nanjing02,Netizen-anywhere,,guangdong,ct&newver=1&newfm=1&secfm=1&flow_ver=3&pkey=14004869e6a6e1dc11ea0835e8b8213bb194cab557d0000003983fda&sl=83034191&expires=8h&rt=sh&r=386249572&mlogid=2748374883349702626&vuk=282335&vbdid=105549401&fin=gcc-arm-none-eabi-4_8-2013q4-20131204-linux.tar.bz2&fn=gcc-arm-none-eabi-4_8-2013q4-20131204-linux.tar.bz2&rtype=1&iv=0&dp-logid=2748374883349702626&dp-callid=0.1.1&hps=1&csl=300&csign=jxOBRm%2B9AOkpJPmKw2etyaLT0P0%3D&by=themis

BaikalHu, 2017-05-02:
Thank you for using gcc compile.
In [doc/LiteOS_Compile_Guide_stm32f429_Gcc.md](https://github.com/LITEOS/LiteOS_Kernel/blob/master/doc/LiteOS_Compile_Guide_stm32f429_Gcc.md), it has a recommended version: gcc version 6.2.1 and higher version.
And you could get the version in this [website](https://launchpad.net/gcc-arm-embedded/+download).

closed 2017-05-02 by BaikalHu

tidyjiang8, 2017-05-02:
Thanks to all of you, I tried just now and it works!

commit 026e4ec1 (2017-05-16): chore(stm32f429): add GNU ARM Embedded Toolchain version detection

add compiler version detection in cmsis_gcc.h

close #65