LiteOS/LiteOS #77: interrupt create failed

RozenAstrayChen, 2017-05-21:
hi i use ==los_api_interrupt.c== but,I got failed in `		uwRet=LOS_HwiCreate(6,0,0,EXTI0_IRQHandler,NULL);` 
i dont know why?
My EVB use STM32F407

RozenAstrayChen, 2017-05-21:
i got error value is `OS_ERRNO_HWI_ALREADY_CREATED`
and it say ** * Solution: Check whether the interrupt specified by the passed-in interrupt number has already been created.
**

zhoubo85, 2017-05-24:
in los_bsp_adapter.c  there is const 
const unsigned char g_use_ram_vect = 0;, 
you should make it to 1, then you can use LOS_HwiCreate

lichao93, 2017-06-05:
IRQ number should add 16 ，if you IRQ number is 6，fill 22 int it。

18959697616, 2017-06-05:
Excuse me，求STMF746ZG 开发板与pc连接的驱动程序

zhoubo85, 2017-06-06:
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link009.html

RozenAstrayChen, 2017-06-07:
OK i got it!

closed 2017-06-07 by RozenAstrayChen