LiteOS/LiteOS #117: errors when i make  liteOS

JSLeo, 2017-08-30:
LiteOS_Kernel]$ make ARCH=arm CPU=cortex-m3 ARMV=armv7-m  PLAT=STM32F429I_DISCO
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s: Assembler messages:
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s:63: Error: selected processor does not support `vpush {s0}' in Thumb mode
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s:64: Error: selected processor does not support `vpop {s0}' in Thumb mode
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s:67: writing to APSR without specifying a bitmask is deprecated
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s:133: Error: selected processor does not support `vstmdb r0!,{d8-d15}' in Thumb mode
kernel/cpu/arm/cortex-m3/los_dispatch_gcc.s:157: Error: selected processor does not support `vldmia r1!,{d8-d15}' in Thumb mode
make: *** [Makefile:260：kernel/cpu/arm/cortex-m3/los_dispatch_gcc.o] 错误 1

BaikalHu, 2017-08-30:
If you choose PLAT=STM32F429I_DISCO, then CPU=cortex-m4, ARMV=armv7e-m.
These values should be paired.

JSLeo, 2017-08-30:
im  sorry,  PLAT=STM32F103ZE-ISO

BaikalHu, 2017-08-30:
Please change ';' to '#'. In gnu style assemble, we use '#' in annotation.

BaikalHu, 2017-08-30:
The make/gcc project is under verify in develop branch.

JSLeo, 2017-08-30:
OK   thank you!

closed 2017-08-30 by JSLeo