LiteOS/LiteOS #127: Translate to english

rprata, 2017-10-25:
Hi, all, 
Is it possible to translate to English docs and README? It's great for the project. 
Best Regards, 
Renan

Foorack, 2019-08-09:
**Against:** English is not a universal language. US tech companies have made it an "unofficial standard", but translating everything into one or more languages is costly and would only slow down the documentation process. Any translated documentation will also quickly be out-of-date whenever the Chinese documentation is updated. There is no problem running the URL through Google Translator.

notinheritable, 2019-08-11:
For: it would be better if it were in English. All source code + comments are in english. The source is also kept on Github which is owned by an American company with millions of English developers working on it. The english docs could easily be translated to Chinese through google translate.

xtlsoft, 2019-08-11:
Did you remember the request for removing all Chinese projects from GitHub Trending?

Because of the special condition of being IT developers (basic English skills, almost all programming languages are in English because of charsets), I think making English the default language is more helpful to build a better open-source community.

This is because of the special IT industry, but not which language is better.

cfrank, 2019-08-15:
For: Huawei is a worldwide enterprise developing an operating system with the intention of being used by people around the world. English is the defacto standard for multinational business dealings and cross-country development. If Huawei wants this project to succeed they will need to provide english documentation at the least and preferably multilingual documentation.

rprata, 2019-08-15:
I've sent a pull request with README translation (#776 ). If maintainers agreed, I'll keep this work for other documents.

**Note: I'm not fluent in Chinese, I'm using google translate and analyzing the source code for this work.**

Jamim, 2019-08-16:
Hello everyone,

> @rprata:
> If maintainers agreed, I'll keep this work for other documents.

Sadly, it looks like there are no maintainers currently.
The only [LiteOS organization](https://github.com/LiteOS)'s repository that can be considered active is [LiteOS_Lab](https://github.com/LiteOS/LiteOS_Lab/branches).

As for me, this whole project doesn't look like a result of work of a giant corporation.

nightrider77, 2021-04-18:
I agree that at least base documentation should be in English. Chinese is a hard language for translator, so translated technical documentation is barely readable... As well as SDK documentations for HiSilicon platform. All these things make impossible to improve software by community. And as we all know, Chinese people have their own vision for software development, which is not so good.