LiteOS/LiteOS #147: mbed-os已足够好了，为什么还要搞这个，世上已有RTX，为什么还要搞这种关中断的内核用于cortex

samos2011, 2018-01-16:
如题

gmphoenix, 2018-05-07:
有没有是一回事，自己能不能控制是一回事，更何况以后会有利益上的纠纷。

gustave-wang, 2018-05-16:
高通已经有手机芯片了，我们还要搞手机芯片干嘛，买就好了。然后。。。

tiiMaker, 2018-05-16:
自主可控。

samos2011, 2018-05-17:
抱歉，我不应该这样说的！但cortex-m上编写os内核是不需要关中断的-------- 原始邮件 --------主题：Re: [LiteOS/LiteOS] mbed-os已足够好了，为什么还要搞这个，世上已有RTX，为什么还要搞这种关中断的内核用于cortex-m (#147)发件人：gustave-wang 收件人：LiteOS/LiteOS 抄送：Sam Lin ,Author 高通已经有手机芯片了，我们还要搞手机芯片干嘛，买就好了。然后。。。

YuanChenmang, 2018-05-23:
liteOS的完成度还是很低的，目前也就是一个调度器而已

blueliuyun, 2018-06-25:
@samos2011 您好，请问该如何理解 “cortex-m上编写os内核是不需要关中断的”，有没有相关资料链接？

samos2011, 2018-06-30:
很多年没有再研究OS了,2012写的cortex-m全功能内核,之后就没再修改过,细节一时想不起来
大致利用cm的几个特性,比如双堆栈,咬尾中断,中断嵌套,SVC调用(次低优先级),pendSV中断(最低优先级)
以及原子指令.
内核函数一般不能重入,所以,唯一的入口是svc调用,当然,在用户中断中调用内核功能是直接调用的,有专用函数,内核函数如果最终需要调度与切换任务,直接悬起pendsv(pendsv处理程序负责调度与切换任务,)而不是在任何地方直接调用,那么中断嵌套完了以后(如发生高优先级中断的话),咬尾中断机制,在最后退出中断时执行pendsv.基本保证的执行顺序而不需要关闭中断
另外,用户中断用到的内核函数可能会被重入,所以这部分函数涉及到的参数的读写需要使用原子指令
差不多了,更多细节可能要慢慢回忆,内核是不需要关闭中断的,用户(外设)中断优先,只需要关注应用设计,内核不会对应用设计有任何影响,但又是多任务环境,这是早期8051,AVR等单片机不可能做到的
通过编写内核,才能感受到cm的精妙之处!

发送自 Windows 10 版邮件应用

发件人: tianyear
发送时间: 2018年6月25日 22:09
收件人: LiteOS/LiteOS
抄送: Sam Lin; Mention
主题: Re: [LiteOS/LiteOS] =?UTF-8?Q?_mbed-os=E5=B7=B2=E8=B6=B3=E5=A4=9F=E5=A5=BD=E4=BA=86=EF=BC=8C=E4=B8=BA=E4=BB=80=E4=B9=88=E8=BF=98=E8=A6=81=E6=90=9E=E8=BF=99=E4=B8=AA=EF=BC=8C=E4=B8=96=E4=B8=8A=E5=B7=B2=E6=9C=89RTX=EF=BC=8C=E4=B8=BA=E4=BB=80=E4=B9=88=E8=BF=98=E8=A6=81=E6=90=9E=E8=BF=99=E7=A7=8D=E5=85=B3=E4=B8=AD=E6=96=AD=E7=9A=84=E5=86=85=E6=A0=B8=E7=94=A8=E4=BA=8Ecortex-m?= (#147)

@samos2011 您好，请问该如何理解 “cortex-m上编写os内核是不需要关中断的”，有没有相关资料链接？

blueliuyun, 2018-06-30:
@samos2011  谢谢您的回复。

closed 2018-10-10 by samos2011