LiteOS/LiteOS #468: 关于cmsis_os2接口标准是不是没有完全实现？

samos2011, 2018-10-11:
今天在对liteos进行C++封装时需要使用cmsis_os2,但发现无法指定额外的控制块，（本人的代码从不依赖具体的内核，仅依赖C++定义的OS接口)。
Attributes structure 中的cb_mem and cb_size成员从未被liteos引用过，应该在头文件注明此功能不可用

    唉！，看起来在osTaskInit使用LOS_MemAlloc函数按宏定义的最大任务数分配内存，其实和静态分配的没有什么区别。还是要在配置文件预计可能的最大任务数，最糟糕的是连mutex等也要事先定义最大数量，无形的增加了上层程序模块间的耦合以及浪费内存。喜欢在定义thread或mutex对象时由静态分配内存或者在构造函数动态分配。
     很不喜欢把动态内存管理与内核绑在一起，无论怎么实现，只有提供线程安全的malloc，free函数才是最方便的。
     从没考虑C++环境，如全局实例的构造函数是在main之前运行的，但构造函数经常的需要使用malloc。

w00164089, 2018-10-11:
实际使用场景是千变万化的，考虑到部分用户可能不希望每次都编译内核，特别是对用户不提供源码只提供二进制动态库和头文件的模组厂商，其最大任务数，mux等信息要求用户控制，这时动态申请是有必要的。

samos2011, 2018-10-11:
mbedos在对CPU硬件资源的抽象，OS的抽象，网络接口抽象做得非常好，建议参考一下。

samos2011, 2018-10-11:
如果要将liteos编译成lib，更应该把它设计成应用无关。

samos2011, 2018-10-11:
最受不了的就是mutex居然要在用户配置文件定义最大限制，mutex是使用最多是部件。mutex的本质就是全局变量，一个mutex对象算上优先级抢占功能撑死32字节。这玩意编译器分配就行了，如同int变量。非要让用户去思考我的项目该分配多少个mutex？

HeMinzhang, 2019-04-29:
我个人的理解（本人非本项目的开发者）:
1、关于复杂性，编译操作系统的用户并不是普通用户。
2、用户有需求
3、一些系统相关值的设置必须指定，难道所有值都让系统去猜测？
4、系统的体积和应用场景。

ppmarc07, 2019-08-17:
> 最受不了的就是mutex居然要在用户配置文件定义最大限制，mutex是使用最多是部件。mutex的本质就是全局变量，一个mutex对象算上优先级抢占功能撑死32字节，会写代码的20字节搞定。这玩意编译器分配就行了，如同int变量。非要让用户去思考我的项目该分配多少个mutex？
> 明显就没有遵守低耦合高内聚

你看FreeRTOS，也是自定义mutex最大数目的，就是用户可自定义啊，便于裁剪