LiteOS/LiteOS #538: What's the reasoning of introduing g_stTskRecyleList?

lynus, 2018-11-02:
I'm not sure why self-deleting task should put the TCB in Recycle list , rather than put it in Free list and deallocate the stack space like the case of deleting other tasks. 
For one reason I can think of is that context switch hook function would check the value of top stack. But this can be worked aroud by pre-allocating a small stack for g_pstTaskCBArray[g_uwTskMaxNum] and asigning to stack top value OS_TASK_MAGIC_WORD.
So I wonder is there any other reason for introducing Recycle list which seems rather redundant to me.

jimscen-wang, 2018-11-05:
Yes, there is another reason to introducing  Recycle list:
 Once task is self-deleting,  the osSchedule must be called, if  the stack deallocate,  the context saving will write illegal memory.

w00164089, 2018-11-05:
Yes, there is another reason to introducing Recycle list:
Once task is self-deleting, the osSchedule must be called, if the stack deallocate, the context saving will write illegal memory.