LiteOS/LiteOS #630: SDK for Huawei Watch GT

alduder, 2018-12-06:
Hi!

Is the LiteOS version in the Huawei Watch GT available to external developers? Is there a SDK for developers to build apps or customise the software for the Watch GT?

Thanks.

azw413, 2019-02-25:
I notice there's no answer yet. You guys have some nice hardware but it's worthless without software. There's probably a few of us quite interested in developing 3rd party watch faces, apps and integrations but we can't do that without any docs, tools or SDK's. How about a simple guide in English covering: how to get hold of the firmware and get it running in QEMU and then how to package it up and deploy changes to the watch ? We can then build the apps whilst you're working out how to distribute them to end users.

vipyoshi, 2019-03-01:
I agree, I want to develop stuff for my watch, kinda disappointed in the software of the Watch GT...
Not even a possibility for developer. Sadly I already read many comment of people who are disappointed in the software :/

xsacha, 2019-03-16:
An SDK for such a watch is unrelated to the kernel the watch uses.
You can likely build apps for Lite OS by targeting this kernel, but the facility to target the GUI and deploy to the watch is unrelated to LiteOS.

As such, this issue should be closed.

blchinezu, 2019-11-04:
Any idea if there will be a way of developing stuff for Huawei Watch GT 2 then?

blopware, 2020-01-28:
Seconded this. If any, Google has a major business success because they allow final users and developers to tweak "their" devices.

If this is not possible with the GT2 (a really good piece of hardware, otherwise) it will eventually fall down.

RafaelPuerta, 2020-06-23:
I want to build an application for me 😥😣

markyin, 2020-07-16:
pls post your question in https://gitee.com/LiteOS/LiteOS

wlad, 2020-11-29:
I know why they dont answer. They dont know english 🤣

xsacha, 2020-11-29:
They don't answer because it is already answered. This is about the OS (LiteOS), which already allows apps. Not the watches, which have no SDK or method of deploying apps.

xsacha, 2020-12-04:
SDK is coming December 12th AFAIK. Here is the docs:
https://developer.harmonyos.com/en/docs/documentation/doc-guides/wearable-developing-0000001052265837

manoj997, 2021-01-03:
@xsacha This is related to HarmonyOS and not LiteOS. The users (developers) of Huawei Watch GT2, GT2e & GT2 Pro cannot be benefited. The reason being no USB or WLAN connectivity options. The only possible way is via Bluetooth. I believe that LiteOS is in its early stages. We have to wait this one out.

xsacha, 2021-01-03:
@manoj997 You'll find that HarmonyOS uses LiteOS (the kernel). https://github.com/Awesome-HarmonyOS/HarmonyOS

As far as I know, the new apps SDK is a streamed system from the phone OS. The apps do not run natively on the watch.

manoj997, 2021-01-03:
@xsacha 
So is it possible to somehow run any custom app? If so, how (steps??)

xsacha, 2021-01-04:
@manoj997 no, of course not :P Although it looks like the GT2 Pro also does not run any apps. It just streams them using an EMUI feature. It requires constant connection to an EMUI-running phone.

manoj997, 2021-01-04:
@xsacha 
I own a Huawei Watch GT2e. LiteOS. This watch is capable of ton of things. I can customize apps and install. Sadly... It's not possible. Any workarounds for this?

nuplo, 2021-01-12:
Any update? I've just bought gt2 pro and need to write an app for the watch

manoj997, 2021-01-13:
@nuplo 
Not yet. :(

NibbleOfAnOctet, 2021-03-11:
Apps are developed using huawei wear engine. Apparently you need an enterprise developer account. 
https://developer.huawei.com/consumer/en/hms/huawei-wearengine/

manoj997, 2021-12-27:
Update: My Huawei GT2e got the firmware update 11.0.14.26 (216 MB) and in the changelog, it says it is possible to install 3rd party apps. 
However, there are some limitations.

- I have a Huawei Nova 3i with Android 9.1 and EMUI 9.0 and in the Huawei health (12.0.12.310) there is no option for installing apps yet.
- Even if you get the install app option for other watches, there is only a handful of apps you can install and it means that you cannot build your own custom app, publish it to AppGallery and install it into your watch.
- With every potential firmware update, the noticable changes are UI changes such as icon colors and updates to weather. Other than that, you can improve things like **Support for Emojis, Support for app icons from the device. (It does not even recognize the Huawei Health App icon), Option to keep notifications even if they're viewed and possiblity to apply quick replies, Answer a call if a earbud is linked to the watch and transfer the call to the watch (This is mainly GT2e), Accuracy of compass and barometer readings, Ability to add custom always on watch faces.**

Dear LiteOS devs, please take a note of this and try to improve it for the betterment.

xsacha, 2021-12-27:
I just got the same update.
You might need a newer app gallery

manoj997, 2021-12-27:
installed AppGallery HUAWEI AppGallery 11.6.1.200_beta and restarted the phone. Still the same :(