LiteOS/LiteOS #735: Always On for Huawei Watch GT

TiTi, 2019-01-07:
It would be really nice to have an option to set display to "**always on**".
It's the only thing missing in this watch for me.

The "Show Time" button only means "Screen always on **for 5 minutes**" (no more)
Basically, it's just when you want to demo. the watch...

The watch light up correctly when i turn my wrist to look at it.
However if my arms rest on the table and I look at the watch, there is only a black screen which is... not ideal for a watch...

Huoke, 2019-01-13:
”always on“ that's not too electricity-intensive。

chaifangming, 2019-02-21:
Hi TiTi,

Thanks for your support of Huawei prodects :-)

But this repo is for LiteOS itself, not for the watch. You can contact customer service.

Best Regards,
Fangming

TiTi, 2019-02-21:
Yeah I contacted customer service but response was deceiving.
Basically it was "You can hope for an update" meaning support did not know anything more than I already did.
It would be nice to know:
* if it's technically feasible?
* is it a known limitation? is it a definitive design choice?
* why don't we have the choice like on other watch platforms?
* if you are working on it to add this ability?

chaifangming, 2019-04-09:
This repo is for LiteOS itself, this is not an LiteOS issue. I closed it for now.

closed 2019-04-09 by chaifangming