LiteOS/LiteOS #760: LiteOS 硬中断接管移植方案

chenshida, 2019-02-22:
我在STM32L476RG板子上移植了LiteOS,根据github上教程移植完系统后,编译运行完全正常,但是教程上采用的是不接管硬中断移植方案.而我的应用场景需要用到LiteOS接管中断,我把
`
#define LOSCFG_PLATFORM_HWI                                 YES
`
这个标志位置为YES来让OS接管硬中断,代码都编译正常,但是运行的时候,只能打印一个字符就不再做任何事情了.我是代码是跑移植里面的example,不采用硬中断接管能够正常打印信息.
github上面有说如果要移植硬中断接管,需要改写启动文件,请问这个文件是los_hw_exc_gcc.S这个文件吗,如果是,我该怎么修改,我把他加入我的工程后编译,会报这样的错误
`
Error: selected processor does not support `vpush {D8-D15}' in Thumb mode
`
我是用cmake组织工程的,用gcc编译工程.
另外,有谁知道官方有提供硬中断接管的移植方案吗,我想自己看看.

chaifangming, 2019-02-25:
您好，抱歉答复迟了，今天刚刚看到您的问题。

要弄清楚中断接管需要改几个文件需要先弄清楚中断接管的工作流程。我这里给您简单介绍一下.

中断接管对硬件的需求：需要MCU支持VTOR寄存器，这样可以更改默认的向量地址，所以cortex-m0肯定不支持。其他MCU一般支持VTOR。

中断接管的工作流程：
1）中断触发后进入我们自己准备的向量表（m_pstHwiForm）中的统一处理函数osInterrupt。
2）osInterrupt函数读取当前irq号
3）m_pstHwiSlaveForm从m_pstHwiSlaveForm数组中取出当前irq对应的处理函数
4）调用这个函数

所以我们所需要的初始化操作就是：
1）在初始化的时候更改vtor寄存器，设置为我们自己的向量表（m_pstHwiForm），而这一步操作在打开（LOSCFG_PLATFORM_HWI = YES）的宏后，会在LOS_KernelInit函数中自当完成。
2）在使能中断的时候将irq处理函数设置到m_pstHwiSlaveForm数组中，并将我们向量表（m_pstHwiForm）中对应当前irq的函数的处理入口改写为osInterrupt，而这两个操作都可以通过一个LOS_HwiCreate函数调用完成。

鉴于VTOR将会被直接更改为我们的数组，您不需要改写自己的汇编代码，只需要做如下更改：
1）定义LOSCFG_PLATFORM_HWI = YES的宏定义
2）使能中断的时候调用LOS_HwiCreate注册

其他不需要更改。los_hw_exc_gcc.S这个文件跟中断接管没有关系的。

您可以参考我们现有的target，现有的target，只要不是m0的，都是中断接管方式的。比如“targets/Cloud_STM32F429IGTx_FIRE”。

如有问题请追问。

chenshida, 2019-02-25:
感谢您的解惑,我明白怎么操作了,万分感谢.

closed 2019-02-25 by chenshida