LiteOS/LiteOS #767: 在文件中los_kifs.c中的函数kifs_file_creat

ifoxhz, 2019-03-28:
if (path_in_mp [strlen (path_in_mp) - 1] == '/')
    {
        return NULL;
    }
如果path_in_mp是0字符串的话，会访问path_in_mp的前面内存，虽然只是读，也许也可以改进一下。

ifoxhz, 2019-03-28:
在同一个函数中：
strncpy (node->name, path_in_mp, t - path_in_mp);
如果进入的字符串是“/huawei/liteos",t-path_in_mp好像是0呀？不知道理解的对不对？

chaifangming, 2019-03-29:
您好，第一个问题可以修改下。

第二个问题，这段代码是用来创建 path_in_mp 中的所有的中间目录的。注意这段代码在一个循环里面。假定您的输入是 path_in_mp = “/huawei/liteos"那么，
在进入下面循环前，path_in_map将成为  path_in_mp = "huawei/liteos"，kifs_file_find会去掉所有的开头"/"，并查找已经创建了的中间目录。这里假定“huawei”这个目录未被创建。
while ((t = strchr (path_in_mp, '/')) != NULL)
这句循环的判断就成功，这个时候 t 为 "/liteos"。t - path_in_mp 刚好是 "huawei"的长度。

ifoxhz, 2019-03-29:
多谢讲解。

郑永

ifoxhz, 2019-03-29:
当时没去看kifs_file_find

郑永
________________________________
发件人: 郑 永 <ifoxhz@hotmail.com>
发送时间: Friday, March 29, 2019 8:57:17 AM
收件人: LiteOS/LiteOS; LiteOS/LiteOS
抄送: Author
主题: Re: [LiteOS/LiteOS] 在文件中los_kifs.c中的函数kifs_file_creat (#767)

多谢讲解。

郑永

closed 2019-04-08 by chaifangming