LiteOS/LiteOS #778: Language

Balls-and-Dick-69, 2019-08-16:
Perhaps a common translation for documentation just go though and try to generalize  what it's saying in English just editing the documentation to be more readable for those interested from a western perspective.

Jamim, 2019-08-16:
Hello the guy with a weird username,

Please don't do it in manner you wrote the sentence above.
Your proposal is not quite readable.

Thanks!

calebxx, 2019-08-20:
what a weird name, what a weird avatar.
I can not understand your issue title

megaacheyounes, 2020-11-29:
Readme.md should be in English as it's the standard language in the tech world, or there should be an English version (README_EN.md) with a link to it in the Chinese README.md file