LiteOS/LiteOS #786: How to try LiteOS on RISC-V

zhoupeng, 2019-09-04:
Hi all,

Does  LiteOS  support any risc-v (rv32 or rv64) board now pls? 
If I try to run LiteOS on risc-v bord, which boards can I choose pls?

I want to get a quick start, how can I run LiteOS on emulator such as qemu-riscv64 pls?

Best,

VincentLee-EN, 2019-09-23:
Did you figure it out? I have the same question with you.

zhoupeng, 2019-09-23:
> Did you figure it out? I have the same question with you.

Not yet, I failed to explore any opensource tool chain or sdk for engaging into system level developing contribution :(

VincentLee-EN, 2019-09-25:
OK, keep going~

advancedwebdeveloper, 2020-01-08:
I am interested to run LiteOS on such boards:
[open-isa.org](open-isa.org)
[longan.sipeed.com/en/](longan.sipeed.com/en/)
.