LiteOS/LiteOS #792: task 里malloc会当机死机,求解决.

panguoxian, 2019-10-19:
平台是：stm32f103
编译平台 ：GCC编译会出现进入HardFault_Handler中断
                    KEIL 编译正常。

触发原因：在creat的task里malloc动态内存。会进入 HardFault_Handler(void) 中断，造成死机。

问题1：在task里使用malloc分配内存是使用creat时分配的uwStackSize大小吗？但我把这个改大了还是一样。

问题2：经常性使用gcc编译会出现不同问题，换成keil编译就正常，请问题使用GCC编译需要注意什么？

iwetuan, 2019-11-07:
比较一下链接脚本的配置信息是不是一样呢

zouyipeng, 2020-08-30:
这种应该是正常使用，可以Share 一下code吗。

bingbingyouni, 2021-01-05:
@panguoxian 请取一份最新的LiteOS最新版代码，试试看。stm32f103已经重新支持了。
编译器版本的话，默认使用版本推荐如下:
LiteOS@ubuntu:~/upai$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.