LiteOS/LiteOS #797: lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

zouzhe1, 2020-03-12:
* **Describe the bug**
lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

* **To Reproduce**
lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

* **Current behavior**
lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

* **Please tell us about your environment:**
lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

* **Other information** (*e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, lite os是烂尾了吗？为啥最近提交是13个月以前的？这一年都没人用？

Ryan3000, 2020-05-22:
没有烂尾，是因为老美的原因，所以已经从github迁出到码云上

Licaifei, 2020-05-22:
[image]
[image]

最近的提交是3天前

suolis-dream, 2020-06-22:
huawei come on !

bingbingyouni, 2021-01-04:
@zouzhe1 
一直都在，欢迎一起搞事情~！有issue继续走起~~
也可以在gitee上找到它~