LiteOS/LiteOS #804: Remove fps limiting on smart watch

Crossmax14, 2020-11-15:
Hey. I would like to offer a function for settings - the ability to remove frame limiting.

I dare to assume that this is a limitation on the part of the software and it was created to increase autonomy.
I think it is necessary to add the ability to turn off this limit, since there are people who have no problems with charging the smart watch every day and they want the interface to work smoothly. Smart watches are also an element of the image.

Most likely this is a specific frame limit (for example, 30 fps) specified in some settings file. I suggest creating a checkbox in the watch autonomy settings so that you can change this limit from 30 fps when the checkbox is on and 60 fps when the checkbox is off. Option text would be "Remove fps limit"

If the developer appreciates this idea positively, as a developer I can try to implement this functionality myself.

bingbingyouni, 2021-01-04:
@Crossmax14  
Have you test this functionality  in LiteOS?

Crossmax14, 2021-07-25:
@bingbingyouni No