LiteOS/LiteOS #805: Graphs

evazzoler, 2020-11-17:
* **Is your feature request related to a problem? Please describe.**
*Barometer graph useless for forecasting*

* **Describe the solution you'd like**
*last 24h graph instead fixed 00-24*

The graph starting from 00:00 and ending at 24:00 is useless, specially for the barometer.
Instant value for a barometric parameter is indicative but almost useless, the most important data is its trend: when pressure encreases, weather becomes better and vice-versa. The faster the pressure drops, the sooner the bad weather arrives. It is possible to predict a local thunderstorm and this is important for safety during trekking, hiking, yachting, etc.
Because of the low speed in barometric variations, the graph should represent at least the last 24h history (would be better if a config let choose the user last 24, 36, 48, 72 with a selectable resolution of 5,15,30,60 min). Just look at every barometer on the other sportwatches (Garmin, Suunto, Casio, etc.) and you will see they are working this way.
With the HUAWEI WATCH GT 2 Pro at 6:00 AM I can see only a 0:00-6:00 graph and this is totally useless for forecasting. I must wait until 24:00 for seeing the last 24h.

This apply to all graphs concepts. Please let the user choose for every graph.

bingbingyouni, 2021-01-04:
I have no idea what your question has to do with liteos?

evazzoler, 2021-03-17:
This is a request for Huawei Wahtches, mainly for HUAWEI WATCH GT 2 Pro.