LiteOS/LiteOS #806: Home Favorites

evazzoler, 2020-11-17:
* **Is your feature request related to a problem? Please describe.**
*User can choose between 6 fixed pre-choosen views, user cannot choose views useful for him*

* **Describe the solution you'd like**
*Let the user choose between every available view*

User can choose as favorites between 6 views (Activity, Weather, Stress, HR, Music, Sleep). User can choose to see 6 or less of them, but not other views. This is no-sense. You could limit to 6 the maximun views (if you have memory limitations, but I think they are'nt), bud you should let the user choose between all the existant views. In example, I would like to set Alarm, Contacts, Call Register, Notifications, Barometer and Compass, if 7 would be permitted, I would add SpO2, too.
On the low button I would put the Camera Remote Trigger, but with the actual concept I must put the barometer there, because the barometer is not selectable as favourite on the home.
Please, let the user add every existant view on favorite, as you did with the low button, and leave - if strictly necessary - the limitation in number and not in type o the favourites.

bingbingyouni, 2021-01-04:
@evazzoler I have no idea what your question has to do with liteos?

evazzoler, 2021-03-17:
Is LiteOS the Operating System running on Huawei Watches?

bingbingyouni, 2021-03-17:
@evazzoler Yes.

evazzoler, 2021-03-17:
Ok, this is a feature request for HUAWEI WATCH GT 2 Pro, which is a Huawei Watch. What's not clear?

bingbingyouni, 2021-03-30:
What is your feature requeset?

evazzoler, 2021-04-05:
User can choose as favorites between 6 views (Activity, Weather, Stress, HR, Music, Sleep). User can choose to see 6 or less of them, but not other views. This is no-sense. You could limit to 6 the maximun views (if you have memory limitations, but I think they are'nt), bud you should let the user choose between all the existant views. In example, I would like to set Alarm, Contacts, Call Register, Notifications, Barometer and Compass, if 7 would be permitted, I would add SpO2, too.
On the low button I would put the Camera Remote Trigger, but with the actual concept I must put the barometer there, because the barometer is not selectable as favourite on the home.
**Please, let the user add every existant view on favorite, as you did with the low button, and leave - if strictly necessary - the limitation in number and not in type o the favourites.**